Poket-Jony/Marlin-A4MaxPro-2.0.x #2: [BUG] Printing after a finished/aborted Print sometimes don't work

Poket-Jony, 2019-12-23:
After some finished/aborted prints i could not print again without restarting.

Additionally i noticed if i cancel a print, the print window with all the information, shows the two main buttons (Print, Resume) in green. If you cancel one more time, they turn blue, then press back once and everything is back to normal.

**Will require further investigation!**

Alexboyru, 2019-12-29:
Confirm. After stop printing, I cannot start a new print.

Poket-Jony, 2020-02-14:
Should be fixed in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/releases/tag/v2.1.0-dev

Poket-Jony, 2020-02-20:
The problem still seems to exist.

Poket-Jony, 2020-02-24:
probably fixed with https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/0ebbe890e3acc8c5d17d8d5463753e769578ffd2

closed 2020-02-24 by Poket-Jony