Poket-Jony/Marlin-A4MaxPro-2.0.x #3: [BUG] Auto-Off not as expected

Erik262, 2019-12-25:
### Bug Description

If you uncheck Auto-Off while printing, the printer nevertheless switches off.

### My Configurations

nothing changed.

### Steps to Reproduce

1. Print something
2. Uncheck Auto-Off in the corner
3. Print finished --> Message appears
4. DON'T click "OK" in the message
5. Wait a few minutes --> Printer goes off. 

**Expected behavior:** No Power off by unchecking in the right corner checkmark

**Actual behavior:** Printer goes off

#### Additional Information

* No M81 or somethingelse is set in the end script.

Frohe Weihnachten ;)

Poket-Jony, 2020-01-02:
Thank you for your contribution.
I'll take a look at that.

Poket-Jony, 2020-02-14:
Fixed in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/releases/tag/v2.1.0-dev
Now the build in marlin function https://marlinfw.org/docs/gcode/M085.html is used for auto power off.

closed 2020-02-14 by Poket-Jony