Poket-Jony/Marlin-A4MaxPro-2.0.x #4: [BUG] Lack of Filament can not be removed

Erik262, 2019-12-26:
### Bug Description

If the printer starts up and the filament sensor recognised that no filament is installed, an Message appears in which you can't remove by clicking the OK button.

### My Configurations

nothing changed 

### Steps to Reproduce

1. Power off printer
2. Remove filament from the filament sensor
3. Turn printer On
4. Message appears and can't be removed --> You have to restart printer an insert some filament to the sensor.

**Expected behavior:** Message should be removed after clicking OK, then continuing with what ever you wanted

**Actual behavior:** Have to restart printer to get off the message

#### Additional Information

Poket-Jony, 2020-01-02:
I can partially confirm that. From time to time I can still click on the OK button.

m-kozlowski, 2020-01-02:
I think that reading FIL_RUNOUT_PIN in `AnycubicTFTClass::Setup()` is redundant.
It's checked anyway as a part of `manage_inactivity`. Checking it on boot is unnecessary and leads to described lockup.

Poket-Jony, 2020-01-03:
Thank you @m-kozlowski 👍 
I have fixed that in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/79c2e1a0ea9bafd4b58c9b0e4a1497290c87d567

closed 2020-01-03 by Poket-Jony