Poket-Jony/Marlin-A4MaxPro-2.0.x #5: [BUG] hotend PID configuration impossible with the suggested sensor

Nestor75, 2019-12-31:
### Bug Description
All my printer is out of the box 

I tried to tunning the hotend PID using M303 S240 C10 with the sensor type 11. It was impossible to get propper values . the temperature shifted +-5 degrees all the time :(

I tried many configurations and at the end what it works for me is:

#define TEMP_SENSOR_0 5//11
#define PID_MAX 150 // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current

  #define PID_FUNCTIONAL_RANGE 15 // If the temperature difference between the target temperature and the actual temperature

  #define DEFAULT_Kp 79.63//22.00//79.63//21.26//18.04//17.27
  #define DEFAULT_Ki 14.95//1.36//13.69//1.40//1.14//0.5//1.05
  #define DEFAULT_Kd 106.00//89.22//115.79//80.98//71.68//80.5//70.73
  
I think it is not a BUG but I want to report it just in case. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Nestor75, 2020-01-02:
I had some problems when using the layer FAN at 100%, the temperature went down more then 16 degrees. 

To avoid that 

  #define PID_FUNCTIONAL_RANGE 5 // If the temperature difference between the target temperature and the actual temperature

in that case it only goes down upto 5-6 degrees when the fan is at 100%.

Poket-Jony, 2020-01-02:
On [this forum](https://drucktipps3d.de/forum/topic/anycubic-4max-pro-marlin-1-1-9-firmware-ai3m-basierend/) alfrank found out that the wrong thermistor type (5) was entered in the stock firmware. The correct one should be 11.
Did you try to do the PID tuning over the "special menu"?

Nestor75, 2020-01-02:
I bought the printer few months ago and Anycubic sent me a new full hotend replacement because the original one was clogged. the printer came also with the carriage XY misaligned... I had to do some fixes...

I tried tunning the PID using the gcode M303 S240 C10, I did tone of test and finally the configuration it worked for me is the one I wrote above. 

What I did not try is with the thermistor type 11 and changing  #define PID_MAX 150 // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
and #define PID_FUNCTIONAL_RANGE 5 // If the temperature difference between the target temperature and the actual temperature

I will try it tomorrow. 

Many thanks :)

Nestor75, 2020-01-10:
I asked the Anycubic Support team and they have confirmed that the correct sensor type is number 5 in marlin:
===================
Hi,

Thank you for your email and support for ANYCUBIC. This is Larry from the ANYCUBIC after-sales technical support team.

You can choose this type of thermistor.
[image]

Good luck.
==================

closed 2020-01-10 by Nestor75

m-kozlowski, 2020-01-10:
well, maybe mine was damaged, but from what I've tested, i had increasing (+20 @170C, +50 @220C) offset between set at measured temp when using sensor type 5.
after changing it to 11 i have constant offset of +10 degree.

Nestor75, 2020-01-10:
This is what support team said but you know.... there is always room for doubting and in the end testing and mesuaring by yourself it what it rules ;)

Nestor75, 2020-01-13:
one question over here, which was the temperature given by the sensor with the wrong marlin code (5) at the stock firmware when you strt the printer?
I mean, in my case with number 5 at marlin, the temperature nowadyas is 15 degress, which is the correct room temperature. and if I try with numer 11 is also 15degrees so...what I dont' know is if these two set ups work similar at low temeratures and they work in a different way when you heat the hot end.

Poket-Jony, 2020-01-17:
I did not test sensor type 5 because alfrank had already done so.
Quote from alfrank:
> I figured it was the wrong type of thermistor because my PETG prints blistered at a nozzle temperature of 230 ° C. So far I have been able to print my PETG perfectly on any printer at 230 ° C. After various tests it came out that the actual temperature is about 20 ° C higher than the set or displayed.

m-kozlowski, 2020-01-17:
> what I dont' know is if these two set ups work similar at low temeratures and they work in a different way when you heat the hot end.

That's the way it works. Both sensor types are NTC type of thermistors, which change resistance based on temperature. That R-T dependency isn't linear. If you'll draw a chart measuring resistances at few known temperature point you'll get a curve. The shape of that curve is described by thermistor beta value. Above sensor types have different beta values: 3950K and 4267K. So as far as they both will give similar readings at ~20C (ADC reads ~986 in both cases) there will be significant difference at higher temps (301 vs 230 @150C, 101 vs 61 @200C)

Nestor75, 2020-01-18:
I am working now using the 11 type like you point and many others also did. I am dealing now with the PID configuration . 
Many thanks for all the info provided