Poket-Jony/Marlin-A4MaxPro-2.0.x #6: [FR] (when the filament sensor triggers stop any kind of print)

Nestor75, 2019-12-31:
### Description

I always print using a raspberry pi with the Repetier server Free conected to the printer via USB.

When the filament sensor triggers the message  is displayed in the screen but the print continues. 
It seems it only works if you print from the SD card. 

I tried to activate the //#define FILAMENT_RUNOUT_SENSOR option but then it gives a compilation error. 

I tried to activate this option in the configuration_adv.h :
#define HOST_ACTION_COMMANDS //nestor activated
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT //nestor uncomented
#endif

but it does nothing when the sensor is triggered. 

I tried to understand what the code does in the marlin.ccp but I cannot find out how to make it works (I am not a coding expert :(  )

Any idea about how to stop the prints when the sensor is triggered even if they are from a USB host?

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Poket-Jony, 2020-01-02:
The current filament failure is handled by the anycubic_TFT class and it does not send the serial command back (see HOST_ACTION_COMMANDS). The best way is to remove the routine for ANYCUBIC_FILAMENT_RUNOUT_SENSOR and replace this with the build in marlin functionality (FILAMENT_RUNOUT_SENSOR) or append the host action commands for a quick fix.

Nestor75, 2020-01-02:
it makes totally sense. 

I will do it and I will  share the outcome :)

many thanks :)

Poket-Jony, 2020-01-02:
Thank you for your participation.
You are also welcome to create a pull request so that we can fix the problem for everyone :-)
I would also leave the ticket open.

Nestor75, 2020-01-02:
I am very good on managing code management systems but I will do my best ;)

Many thanks for your help

Poket-Jony, 2020-01-17:
Big thanks. Good work 👍 
I look at that and import it into GitHub.

Nestor75, 2020-01-18:
some comments, 

1- the steps per mm are the original ones and as per my tests are the good ones
2- the pid configuration have to be tonnued properly, I am still working in that and I will put my finding later.
3- the s curve is disabled as I had problmes with Y axis shifting layers

I think that's all I will update the files I published as soon as I got a good final conf

Nestor75, 2020-01-18:
//===========================================================================
//============================= PID Settings ================================
//===========================================================================
// PID Tuning Guide here: http://reprap.org/wiki/PID_Tuning

// Comment the following line to disable PID and enable bang-bang.
#define PIDTEMP
#define BANG_MAX 255   //nestor  // Limits current to nozzle while in bang-bang mode; 255=full current
//nestor
//#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#define PID_MAX 150 //nestor Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#define PID_K1 0.80//1//0.95     // Smoothing factor within any PID loop
#if ENABLED(PIDTEMP)
  //#define PID_EDIT_MENU         // Add PID editing to the "Advanced Settings" menu. (~700 bytes of PROGMEM)
  //#define PID_AUTOTUNE_MENU     // Add PID auto-tuning to the "Advanced Settings" menu. (~250 bytes of PROGMEM)
  //#define PID_DEBUG             // Sends debug data to the serial port.
  //#define PID_OPENLOOP 1        // Puts PID in open loop. M104/M140 sets the output power from 0 to PID_MAX
  //#define SLOW_PWM_HEATERS      // PWM with very low frequency (roughly 0.125Hz=8s) and minimum state time of approximately 1s useful for heaters driven by a relay
  //#define PID_PARAMS_PER_HOTEND // Uses separate PID parameters for each extruder (useful for mismatched extruders)
                                  // Set/get with gcode: M301 E[extruder number, 0-2]
  #define PID_FUNCTIONAL_RANGE 10//5//15 //nestor // If the temperature difference between the target temperature and the actual temperature
                                  // is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.

  // If you are using a pre-configured hotend then you can use one of the value sets by uncommenting it
  //nestor

// bueno con pid max 150 10degrees -> ok pid k1 =0.8
#define DEFAULT_Kp 15.11
#define DEFAULT_Ki 0.94
#define DEFAULT_Kd 60.97

Poket-Jony, 2020-01-21:
I will rewrite the `anycubic_TFT` class to work out of the box with the `runout` class in Marlin.
See https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/issues/11

closed 2020-01-21 by Poket-Jony