Poket-Jony/Marlin-A4MaxPro-2.0.x #11: [FR] Rewrite whole display class

Poket-Jony, 2020-01-20:
Marlin 2.0 offers the possibility to program a display with `extensible_ui`.
Therefore, the current `anycubic_TFT.cpp` code should be completely rewritten and adapted to the `ExtUI` class.
This makes the code more readable and the native Marlin functions like filament runout are supported.

Poket-Jony, 2020-01-25:
## Status update
https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/2bf63348d07ef02d4f131aba8b8f85c4cae1c955 is ready for testing.
The special menu is currently removed and will be added in the future.

**Attention: The build is untested! Some parts may not work yet!**

Poket-Jony, 2020-02-13:
## Released v2.1.0-dev
https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/releases/tag/v2.1.0-dev

bernhardberger, 2020-02-14:
I'll beta test this.

You're the single reason why I haven't ditched the Trigorilla Board yet.

Is this targeted against the latest Marlin 2.0.x? Just asking because I'm thinking of ditching the Allegro Extruder driver and the latest Marlin has some fixes for the TMC in it..

Poket-Jony, 2020-02-14:
I merged the upstream branch "[bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)" yesterday. So the version is currently based on the Commit [#62d7d](https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/62d7d49719d3726fff7039d7185f3215e1517db4) from Marlin.

Poket-Jony, 2020-02-17:
First RC-Release: https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/releases/tag/v2.1.2-rc

closed 2020-02-17 by Poket-Jony

TelliMantelli, 2020-02-25:
Testing this also!