Poket-Jony/Marlin-A4MaxPro-2.0.x #18: [BUG] Wrong Heatbed Thermistor

bernhardberger, 2020-02-15:
According to the DT3D board the Bed Thermistor should be `Type5` (not verified by me)

Poket-Jony, 2020-02-17:
Merged in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/pull/19

closed 2020-02-17 by Poket-Jony

Nestor75, 2020-02-17:
I would say that the sensor type is 11 instead of 5.
The reason are, the anycubic official store says the sensor is Sensor de calor: 100 K 1% B = 3950
so the code in marlin is 11.
And the another reason is, I use type 11 for both sensors, hotend and bed becasue I read it in many places the sensors  are the same and I verified the good one for hotend is 11. 

I would say the correct code is 11 .

bernhardberger, 2020-02-17:
Good question.. someone tore the Ultrabase for the 4MP apart recently and meant it's (actually) Type5 on the heatbed, not 11..

Nestor75, 2020-02-17:
the point is that code 5 and code 11 give more or less the same temparature at low degrees. For example, at room temerature both codes give exactly the same temparature. The problem is more relevant wuth the hotend becasue with code 5 it gives 230 degres but the real temeprature could be above 250 degres. 
the hotbed doesn't go above 110 degrees so..... the most probably thing is the code here does not affects almost anything.

PR #19 (2020-02-17): [BUGFIX] Use Type5 for bed temp sensor
fixes #18