Poket-Jony/Marlin-A4MaxPro-2.0.x #24: [BUG] Display Filelist does not work correctly

Poket-Jony, 2020-02-20:
**Version: 2.1.2-rc**

Big problem: the file selection in the print menu doesn't work at all. Only 11 of 11 files are displayed, one of them double.

After various tests, I reduced everything to a minimum: the SanDisk 8 GB is freshly formatted in FAT32 mode, the file names consist of only one character (lowercase letter or number) (plus the extension .gcode), they are located directly in the main directory. The mistake remains.

Depending on the content of the SD card, something else is displayed, relatively randomly ...

**I suspect that the control from the display is still wrong or the entries are transferred too quickly.**

[image]
[image]

Poket-Jony, 2020-02-20:
currently not reproducible

Poket-Jony, 2020-02-21:
bug found: the elements are displayed twice on the first page because the maximum limit is incorrectly calculated

Poket-Jony, 2020-02-21:
fixed in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/a4905c771d3a0c74ff854a71c6fbdfde91586197

closed 2020-02-21 by Poket-Jony

TelliMantelli, 2020-02-26:
using A4MaxPro-2.0.x-2.1.2-rc also getting double file name. Only one gcode file on SD listed twice on LCD.