Poket-Jony/Marlin-A4MaxPro-2.0.x #30: [BUG] Autotuning Hotend / Hotbed in special menu not working

achrainer, 2020-02-24:
Hi,

it seems that with the latest release, the Auto-Tune PID menu items do no longer work. The machine does not heat up anything and turns off within 1-2 minutes.

Best regards from Austria,
Gerhard

Poket-Jony, 2020-02-24:
Please connect to the printer via Pronterface (or similar) and have a look at the output. There should be exactly what he does.

achrainer, 2020-02-24:
Servus,

schau ich mir natürlich mal an, kein Problem. Mir ist’s nur aufgefallen, weil die ersten 2.x Versionen hier kein Problem machten. Dachte, es wäre evtl. das gesamte Special Menu nicht in Ordnung.

Wenn’s Erkenntnisse gibt, aktualisiere ich das Ticket.

LG Gerhard

(English)
Ok, I’ll have a look and update you if there is more detail

Poket-Jony, 2020-02-24:
In the last RC versions, the new shutdown M85 was added, which switches the printer off after 15 minutes of inactivity. May be that you did the PID tuning right then. It is best to deactivate "auto power off" in the status menu before do the PID Tuning.

bernhardberger, 2020-02-24:
Works for me. Only a missing "peep"-Tone on the bed pid tuning, but worked like a charm.

Poket-Jony, 2020-02-25:
https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/blob/master/Marlin/src/lcd/extensible_ui/lib/dwin/DwinTFT.cpp#L314 should produce the sound .. but the method may not be called

closed 2020-02-28 by Poket-Jony