Poket-Jony/Marlin-A4MaxPro-2.0.x #31: [BUG] (short description)

Erik262, 2020-02-26:
### Bug Description

While I was making a mesh level with my BL-Touch 3.1v. The screws behind the printhead case was hitting the Z-axis thread (the left one if the printer is looking in front)

Maybe this could also caused by a non BL-Touch using person.

### Steps to Reproduce

1. Go to the upper left corner (X min and Y max)
2. Go to X Max
3. Screws of the printer head case will hit the Z-axis thread

#### Additional Information

In the configuration file I found this:
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -8.5
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 272
#define Y_MAX_POS 215
#define Z_MAX_POS 210

In the previous version the code was like this:

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -8
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 205

-----

This solved my problem
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE

Poket-Jony, 2020-02-26:
the faulty pull request: https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/4033a819e870ccc4d30d8737a049b6dccf7bd395

I will fix this

Poket-Jony, 2020-02-26:
fixed in https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/commit/158709bd6ca4b7ce5be380f422bc54c7b618de1d

closed 2020-02-26 by Poket-Jony