Poket-Jony/Marlin-A4MaxPro-2.0.x #41: [BUG] Constant reboot

MatLeb, 2020-04-26:
Hi, thanks a lot for this firmware customization !

### Bug Description

My A4MaxPro keeps rebooting.

### My Configurations

No modification in `Configuration.h` and `Configuration_adv.h`
[Configuration.zip](https://github.com/Poket-Jony/Marlin-A4MaxPro-2.0.x/files/4535669/Configuration.zip)

### Steps to Reproduce

1. Update de firmware
2. Boot start
3. Printer reboots immediately after a few lines (last line : `Filament runout sensor: M412 S1`)

#### Additional Information

* Log 

_Recv: start
Recv: echo:Marlin 2.0.0
Recv:
Recv: echo: Last Updated: 2020-03-07 | Author: (Poket-Jony, Github)
Recv: echo:Compiled: Apr 26 2020
Recv: echo: Free Memory: 2166 PlannerBufferBytes: 1568
Recv: TFT Serial Debug: statusChanged: Anycubic 4Max Pro Ready.
Recv: echo:SD card ok
Recv: echo:V76 stored settings retrieved (747 bytes; crc 59879)
Recv: echo: G21 ; Units in mm (mm)
Recv:
Recv: echo:; Filament settings: Disabled
Recv: echo: M200 D1.75
Recv: echo: M200 D0
Recv: echo:; Steps per unit:
Recv: echo: M92 X100.00 Y80.00 Z800.00 E418.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo: M203 X150.00 Y150.00 Z20.00 E80.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo: M201 X700.00 Y700.00 Z70.00 E15000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo: M204 P700.00 R700.00 T700.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo: M205 B20000.00 S0.00 T0.00 J0.04
Recv: echo:; Home offset:
Recv: echo: M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo: M420 S0 Z0.00
Recv: echo:; PID settings:
Recv: echo: M301 P18.53 I1.27 D67.55
Recv: echo: M304 P100.68 I17.07 D395.86
Recv: echo:; Linear Advance:
Recv: echo: M900 K0.00
Recv: echo:; Filament load/unload lengths:
Recv: echo: M603 L100.00 U100.00
Recv: echo:; Filament runout sensor:
Recv: echo: M412 S1_

Erik262, 2020-04-26:
Did you send these commands (M502, M501, M500) between step 1 and 2?

MatLeb, 2020-04-26:
I can't : the firmware reboot automatically after update (maybe I'm doing something wrong) and after this reboot it's not possible anymore to send commands.

Erik262, 2020-04-26:
I guess that something got wrong while flashing the firmware. Did you use the precompiled hex file or you made it yourself? If nothing helped, try this: https://www.youtube.com/watch?v=HahGaOTUmUw

MatLeb, 2020-04-26:
Ok thanks !
In fact I did both (precompiled hex + compiled from sources), exactly the same result.
When I flash it back to the original firmware Anycubic 1.1.7 (Marlin 1.1.0-RC8), no problem.
Will try to do it from cura (I used platformIO on Mac) and check the jumper setting on the board, to see if it changes something.

Erik262, 2020-04-27:
Okay, that's interesting. I'm also compiling it by myself on my Mac with platformio. Everything works. And you can't see any errors?

MatLeb, 2020-04-27:
Yes, no error, only 2 small warnings during compile :

> In file included from Marlin/src/gcode/temperature/../../inc/../core/boards.h:24:0,
                 from Marlin/src/gcode/temperature/../../inc/MarlinConfigPre.h:35,
                 from Marlin/src/gcode/temperature/../../inc/MarlinConfig.h:28,
                 from Marlin/src/gcode/temperature/M888.cpp:23:
Marlin/src/gcode/temperature/../../inc/../core/macros.h: In instantiation of 'constexpr decltype ((lhs + rhs)) _MAX(L, R) [with L = unsigned int; R = int; decltype ((lhs + rhs)) = unsigned int]':
Marlin/src/gcode/temperature/M888.cpp:41:69:   required from here
Marlin/src/gcode/temperature/../../inc/../core/macros.h:300:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         return lhs > rhs ? lhs : rhs;

And

> In file included from .pio/libdeps/mega2560/LiquidCrystal_ID136/LiquidCrystal_SR1W.cpp:48:0:
.pio/libdeps/mega2560/LiquidCrystal_ID136/LiquidCrystal_SR1W.h:168:1: warning: multi-line comment [-Wcomment]
 //             |    |      0.1uF            |     \

Qbalion, 2020-06-13:
It seems that such behaviour occurs when you update while having SD card inserted. Try to put SD card out and then rebooting stops to happen.

coolflyreg, 2021-02-06:
I have the same problem.
It always reboot after uploaded firmware.
I used history version (e593da1c23a4759f6591826f815a504f0272e21a), it works fine.
But it can't works on last version.

coolflyreg, 2021-02-06:
@MatLeb 
I found another solve solution: 
To enable LONG_FILENAME_HOST_SUPPORT, and disable SDCARD_SORT_ALPHA, SDCARD_RATHERRECENTFIRST.
Might be the sort file function caused out of memory.

mr-anderson-se, 2021-11-16:
I had updated firmware to the v2.1.9-rc (Marlin 2.0.x) and printer kept rebooted after it.
After I had put out SD card it booted normally. I had to format SD to make it working with printer again.