Poket-Jony/Marlin-A4MaxPro-2.0.x #47: Manual Bed Mesh Leveling/ Not work

sambroprzem, 2020-11-24:
Hi,
Manual bed leveling does not work. On the SD card, manual leveling is available, the printer levels all 25 points. Unfortunately, at the end (after 20 seconds), nothing happens after saving to EPPROM. The printer is not making a special noise. In the Cura start code I have (M501 and M420 S1). The printer does not use saved leveling points :(.
Where to look for the cause?
thanks

sambroprzem, 2020-12-12:
Does anyone read this and can help?

Erik262, 2020-12-12:
To save your Mesh in your EPPROM you have to do a M500 command followed by M420 S1 as you mentioned before. M501 use used for reporting your settings. Probably your problem is there?

sambroprzem, 2020-12-13:
Thank you.  
The M501 and M420 S1 commands are recommended by Pocket Johny.  Do you think we should change to the M500 and M420 S1.  I save the grid to Eproom via the LCD screen, is it supposed to save it somehow, e.g. via Octoprint or Repetier Host (this is how I connect the printer to the computer)?  
What do you think?
Thanks