Poket-Jony/Marlin-A4MaxPro-2.0.x #48: [FR] Anycubic 4Max Pro v2 support

me21, 2021-03-23:
Hello,

Anycubic 4Max Pro v2 is out. Is it tricky to add its support?

superrootplus, 2021-03-23:
Hello,

Anycubic 4Max Pro v2 is out. Is it tricky to add its support?

Erik262, 2021-03-23:
It is not so easy, because you also have to adjust the Screen's firmware, since it is using a different board as I know.

YairHH, 2021-05-20:
Hi, 

I control my machine with octoprint, I just bought a 4 max pro v2 < if I install this firmware; will it work? if the screen does not work, could I continue to just operate thru octoprint? Thanks.

Erik262, 2021-05-20:
No this won't work, since 4Max Pro V2 is a different machine as the V1

lijgame, 2021-05-20:
there is a good chance that 4Max Pro 2.0 already runs on Marlin 2, so you don't need to flash this firmware.

JonnyHaystack, 2021-07-29:
@lijgame nah it still uses a really old version hacked together with the wrong thermistor set in the code... Anycubic's firmware is pretty pathetic. There is a custom 2.0 build here though https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/65

knutwurst, 2021-10-30:
I've included the 4MAX pro 2.0 build environment for stock and TMC. I actually don't know if all motor directions are correct, so please give me some feedback, what's not working correctly
[KW-131-4MAXPRO2.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7446174/KW-131-4MAXPRO2.zip)
.