Poket-Jony/Marlin-A4MaxPro-2.0.x #50: BLTOUCH pin

kennethgomez01, 2021-09-29:
How to wire BLTOUCH to anycubic 4 max pro?

Erik262, 2021-09-29:
You can use this...

[image]

kennethgomez01, 2021-09-29:
@Erik262 Hi, is this your printer setup? If it is, can you show me how you put the BLTouch sensor on the hot end? And please tell me the pin of where to plug the bltouch sensor? Thanks.

Erik262, 2021-09-29:
I put my BLTouch sensor inside of the head case. The wiring is exactly the same as the "I3Mega". Take a closer look on the picture, my connectors al labled like this one: 
Cable Color (GRB) where is 'G = +' and 'R = -' etc.

kennethgomez01, 2021-10-24:
on your repo its causing error `ui_api 2.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

kennethgomez01, 2021-10-30:
Hi I am just wondering as I am having toruble with z-offser(value is negative) for few days, just want to know is there z offset wizzard for anycbic 4 max pro?

kennethgomez01, 2021-10-30:
@Erik262  Hi I am just wondering as I am having toruble with z-offser(value is negative) for few days, just want to know is there z offset wizzard for anycbic 4 max pro?

Erik262, 2021-10-30:
As far as I know, there is no z-offest wizard in that Marlin version.

kennethgomez01, 2021-10-30:
@Erik262 just question did you remove the z-max switch for bltouch? Because I followed every totorial about z-offset but not be able to succed. I am getting bed not sticking

Erik262, 2021-10-30:
No, I didn't, but if it is touching before the BLTouch does, then you have to move it up a bit. You might not save it correctly in your EEPROM or something else.
Try to set the z-offset manually, by using a feeler gauge (no paper!) and the wheelknobs under your heatbed. Then do your BLTouch stuff.

kennethgomez01, 2021-10-31:
@Erik262 can you take a pciture of your setup of bltouch attach to the extruder?

Erik262, 2021-10-31:
@kennethgomez01 Here are some pictures of my setup on Thingiverse: https://www.thingiverse.com/thing:4090425

kennethgomez01, 2021-11-02:
@Erik262 are you using the clone? Why I keep getting print not sticking in the bed in some area of the bed? Can you share me the slicer setting you use, and grid points?

Erik262, 2021-11-02:
I used to use a clone. Now I'm using the original one. The reason why your prints are not sticking on the bed is probably that your z-offset is not correct or you have some weird concave areas on your printbed. You might have to make a fine grid. I think I set in the settings to 25 points. I'm not sure. Or use a gluestick :)

kennethgomez01, 2021-11-02:
@Erik262 how do you manage to set z_offset? Mine I change the hot end to v6 but the heatsink is same.

Erik262, 2021-11-02:
You have to set up your z-offset only once like you wouldn't have a BLTouch. Then correct your z-offset with your BLTouch. If you have the wrong z-off set, you will provide this at any place on your printbed. You can do this by using the code M-851 or something else. This should be the easiest part. It's been a few years ago since I set this on my printer. I don't know everything what I did there.

kennethgomez01, 2021-11-02:
Thanks @Erik262